Billiam/cncjs-pendant-streamdeck #1: Request: Spindle control (on/off/speed)

BrashEndeavours, 2022-07-30:


Billiam, 2022-07-30:
I believe this functionality is already available:

Spindle on/off/speed can be accomplished with the [gcode](https://billiam.github.io/cncjs-pendant-streamdeck/docs/actions/#gcode) action (`M5` to stop the spindle, `M3` for clockwise full on, and `M3 S1500` for 1500 RPM, and `M4`/`M4 S<speed>` for counterclockwise).

You can also incrementally override the speed with the [spindle](https://billiam.github.io/cncjs-pendant-streamdeck/docs/actions/#spindle) and [toggleSpindleInterval](https://billiam.github.io/cncjs-pendant-streamdeck/docs/actions/#togglespindleinterval) actions.

Does that help?

BrashEndeavours, 2022-07-30:
Oh I mean more of a default button on the jog menu.

Billiam, 2022-07-30:
I see!

The example is really just there to get started (and not everybody will need everything), but you should be able to add anything you need there :)

closed 2022-07-30 by Billiam