Billiam/cncjs-pendant-streamdeck #6: CNC position updates are cut off

BrashEndeavours, 2022-08-11:
The CNC position status screen text is truncated. Adjusting the settings.json font size only seems to effect the web UI.

[image]

Billiam, 2022-08-11:
The stream deck process doesn't do any automatic font scaling or anything, so it's kind of up to you to either use small text or carefully curate the text content itself to keep it's width/height to a minimum for your needs.

Are you sure your font size setting isn't affecting the stream deck display? If so, that's a bug. Note that you'll need to restart the process after making any configuration changes.

My personal preferences is to use a smaller pixel font, which looks like this: 
[image]

My config uses:

```json
"ui": {
    "font": "ProggyTinyTT",
    "fontSize": 16,
    "lineHeight": 0.75,
}
```
from [proggy fonts](http://www.proggyfonts.net/index.php?menu=download)

Billiam, 2022-08-11:
I'm planning (at some point) to add a general override for the UI config section that would only affect the stream deck to make sharing a single config a bit easier.

BrashEndeavours, 2022-08-11:
My fault. It was the line height!
Thank you for sharing your configuration.

closed 2022-08-11 by BrashEndeavours