Billiam/cncjs-pendant-streamdeck #7: "if" fonction seem to not work....

iconmix, 2022-08-15:
hello.....
this is my configuration :
CNC homemade + arduino UNO + raspberry pi 3B+ with 7" LCD HDMI TOUCH screen
CNCJS work perfectly on RPi.
MachinePosition button never showing :(
[image]

"machinePosition": {
      "actions": [
        {
          "action": "toggleUserFlag",
          "arguments": ["showAbsolutePosition"]
        }
      ],
      "bgColor": 8,
      "if": "ui.userFlags.showAbsolutePosition !== true",
      "text": "--{{cnc.workflowState}}--\nMachine Pos\n{{cnc.displayMpos}}",
      "textAlignment": "left"
    },

I've tried many condition like "cnc.idle","cnc.alarm","ui.web" but "if" block always the button....
with "if" : "cnc.idle" 
If i remove "if" line, the button appears and i can see "cnc.idle" is true....
[image]

So i think the "if" function don't work for me :(

Any clue about this ?

And, under windows or linux, how to build your package ? i've installed nodejs, vue, vite but always error at build :(
I'm sure i don't do the right thing .....
Thanks in advance

Billiam, 2022-08-15:
There's an issue with the build system for the web version that I think is affecting you here.

I believe I have a fix for it that I'll be pushing out soon

iconmix, 2022-08-15:
> There's an issue with the build system for the web version that I think is affecting you here.
> 
> I believe I have a fix for it that I'll be pushing out soon

Thank you....
I'm  a noob in nodejs dev :(  sorry

Billiam, 2022-08-17:
This should be fixed in [0.2.5](https://github.com/Billiam/cncjs-pendant-streamdeck/releases/tag/0.2.5)

closed 2022-08-17 by Billiam