Billiam/cncjs-pendant-streamdeck #8: cncjs "command"

iconmix, 2022-08-24:
Hi...
definitively the best pendant for cncjs, streamdeck can call cncjs command ?
in cncjs, you can definite your own command like that : 

`"commands": [
        {
            "title": "Update (root user)",
            "commands": "sudo npm install -g cncjs@latest --unsafe-perm; pkill -f cncjs"
        },
        {
            "title": "Update (non-root user)",
            "commands": "npm install -g cncjs@latest; pkill -f cncjs"
        },
        {
            "title": "Reboot",
            "commands": "sudo /sbin/reboot"
        },
        {
            "title": "Shutdown",
            "commands": "sudo /sbin/shutdown"
        }
    ],`

But i can't find how to call from streamdeck button ....
Is possible or not ?
Thanks for your great job !!!

Billiam, 2022-08-24:
This isn't possible in the current version, but I'll see about getting it added.

Billiam, 2022-08-26:
Support for commands added in [0.4.1](https://github.com/Billiam/cncjs-pendant-streamdeck/releases/tag/0.4.1)

Documentation for this action can be found here: https://billiam.github.io/cncjs-pendant-streamdeck/docs/actions/#command

closed 2022-08-26 by Billiam