Billiam/cncjs-pendant-streamdeck #9: filelist seem got problem

iconmix, 2022-08-27:
Hello....
Thanks for "command" action .....really nice !!!!!

I've a problem with filelist .......
first screen :
[image]

then i select gcodes directory who contain 21 files and as you can see the down arrow is not present....:( : 
[image]

i click on Back button and return by clicking on filelist button (loadGcode) : 
[image]

Now i get down arrow but if i click on it, nothing change execpt the 2nd button (yellow circle on picture)
[image]

It seem not refeshing all buttons.....
Any clue ?

And thanks again for your excellent work !!!!

Billiam, 2022-08-27:
That doesn't look right :smile:

What browser are you using? What operating system?

Do you see any errors in the javascript console?

Is your configuration [valid](https://billiam.github.io/cncjs-pendant-streamdeck/validator/)?

iconmix, 2022-08-27:
hi...
my config : 
_Raspberry pi 4 on chromium brower with 7inch screen_ 
and 
_PC Windows 10 with firefox.....(same problem)_

console output (sorry i'm french ;) ) : 

`unreachable code after return statement index.js:14496:6
Connected to port /dev/ttyACM0 index.js:17756:15
unreachable code after return statement index.js:14496:6
Certains cookies utilisent incorrectement l’attribut recommandé « SameSite » 3
Le cookie « lang » n’a pas de valeur d’attribut « SameSite » appropriée. Bientôt, les cookies dont l’attribut « SameSite » est manquant ou défini avec une valeur invalide seront traités comme « Lax ». Cela signifie que le cookie ne sera plus envoyé dans des contextes tiers. Si votre application dépend de la disponibilité de ce cookie dans de tels contextes, veuillez lui ajouter l’attribut « SameSite=None ». Pour en savoir plus sur l’attribut « SameSite », consultez https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite files
Le cookie « lang » n’a pas de valeur d’attribut « SameSite » appropriée. Bientôt, les cookies dont l’attribut « SameSite » est manquant ou défini avec une valeur invalide seront traités comme « Lax ». Cela signifie que le cookie ne sera plus envoyé dans des contextes tiers. Si votre application dépend de la disponibilité de ce cookie dans de tels contextes, veuillez lui ajouter l’attribut « SameSite=None ». Pour en savoir plus sur l’attribut « SameSite », consultez https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite files
Le cookie « lang » n’a pas de valeur d’attribut « SameSite » appropriée. Bientôt, les cookies dont l’attribut « SameSite » est manquant ou défini avec une valeur invalide seront traités comme « Lax ». Cela signifie que le cookie ne sera plus envoyé dans des contextes tiers. Si votre application dépend de la disponibilité de ce cookie dans de tels contextes, veuillez lui ajouter l’attribut « SameSite=None ». Pour en savoir plus sur l’attribut « SameSite », consultez https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite 2 files
`

Billiam, 2022-08-27:
Thanks! This has been fixed in [0.4.3](https://github.com/Billiam/cncjs-pendant-streamdeck/releases/tag/0.4.3)

closed 2022-08-27 by Billiam

iconmix, 2022-08-28:
big thanks ! Now it's perfect :)