Billiam/cncjs-pendant-streamdeck #13: Pendant not running

crackinx, 2023-08-30:
trying to figure out my issue. I've got everything set up correctly to the best of my knowledge, cncjs runs and says the pendant folder is mounted to serve requests for grid but when I go to my cncjs ip:8000/grid/ I get a page not found error. obviously I am trying to get the web interface running not the stream deck extension. NO clue what the issue could be, any suggestions?

Billiam, 2023-08-30:
Since the pendant is just html+js, it'll be down to your cncjs setup and configuration unfortunately.

Did you follow the instructions in the readme? https://github.com/Billiam/cncjs-pendant-streamdeck#installation

What does the mountPoints section of your`.cncrc` file look like? Where do you have cncjs-pendant-streamdeck installed?

crackinx, 2023-08-30:
I followed the directions exactly, so everything is as the directions say including directory name and location.

Billiam, 2023-08-30:
Are you able to mount other pendants, like https://github.com/cncjs/cncjs-shopfloor-tablet ?

Do you see any logs related to mounting the CNC js log file? You may need to increase log verbosity. You should see something like:   `Starting a proxy server to proxy all requests starting with 'grid' to 'some path'`, or similar. Do those paths look correct?

Any other errors that look relevant?

Less likely on raspberry pis, but are there any permission issues that would prevent the cncjs process from serving those files?

Other than that, I don't  really have any other suggestions

crackinx, 2023-08-30:
You know, permissions are a good place to start, I am running on wsl,so there might be a folder permission issue.

crackinx, 2023-08-30:
also yes, I can load the tablet pendant just fine.

crackinx, 2023-09-01:
so, I have made no progress on getting the pendant to run, log shows no issues, other pendants run without any problems. I'm at a loss.

crackinx, 2023-09-01:
this is what I get on the log with -vvv, all permissions are correct. it's very odd, since other pendants will run just fine. I'll keep at it.

`2023-09-01T16:53:03.309Z - info init Loading configuration from "/home/ynot/.cncrc"     at Object.createServer (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:72:7)
2023-09-01T16:53:03.311Z - info init Mounting a directory "/home/ynot/cncjs-pendant-streamdeck" to serve requests starting with grid     at /usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:207:11
2023-09-01T16:53:03.346Z - debug app app.settings: {"x-powered-by":false,"etag":"weak","env":"production","query parser":"extended","subdomain offset":2,"trust proxy":true,"views":["/usr/lib/node_modules/cncjs/dist/cncjs/app","/usr/lib/node_modules/cncjs/dist/cncjs/server/views"],"jsonp callback name":"callback","view cache":true,"verbose errors":false,"case sensitive routing":true,"strict routing":false,"view engine":"html"}     at appMain (/usr/lib/node_modules/cncjs/dist/cncjs/server/app.js:101:9)
2023-09-01T16:53:03.369Z - debug app assets: name=app, asset={"routes":["/6fc6c920/","/"],"path":"/usr/lib/node_modules/cncjs/dist/cncjs/app","maxAge":31536000000}     at /usr/lib/node_modules/cncjs/dist/cncjs/server/app.js:206:9
2023-09-01T16:53:03.369Z - debug app > route=app /6fc6c920/     at /usr/lib/node_modules/cncjs/dist/cncjs/server/app.js:213:11
2023-09-01T16:53:03.370Z - debug app > route=app /     at /usr/lib/node_modules/cncjs/dist/cncjs/server/app.js:213:11
(node:1312) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
2023-09-01T16:53:03.376Z - error init Error: listen EADDRINUSE: address already in use 0.0.0.0:8000
    at Server.setupListenHandle [as _listen2] (node:net:1872:16)
    at listenInCluster (node:net:1920:12)
    at doListen (node:net:2069:7)
    at process.processTicksAndRejections (node:internal/process/task_queues:83:21) {
  code: 'EADDRINUSE',
  errno: -98,
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000
}     at EventEmitter.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:269:9)
Error: Error: listen EADDRINUSE: address already in use 0.0.0.0:8000
    at Server.setupListenHandle [as _listen2] (node:net:1872:16)
    at listenInCluster (node:net:1920:12)
    at doListen (node:net:2069:7)
    at process.processTicksAndRejections (node:internal/process/task_queues:83:21) {
  code: 'EADDRINUSE',
  errno: -98,
  syscall: 'listen',
  address: '0.0.0.0',
  port: 8000
}
^[[23~`

Billiam, 2023-09-01:
Do you have logs for the 404? 

Do you have some other server in front of cncjs (apache, nginx, some kind of proxy), that might prevent those requests from passing through to cncjs?

crackinx, 2023-09-01:
nope no server in front, not sure where to even look for the 404 logs

crackinx, 2023-09-01:
progress!! the mount point was pointing to grid instead of /grid. I now get a loading screen that it is stuck on, but that might be the config

crackinx, 2023-09-01:
Success!! fully running and functional!

Billiam, 2023-09-01:
Thanks for your patience, I'm glad it's working!

commit 003aee90 (2024-12-07): Improve setup docs

Fix #13
closed 2023-09-01 by Billiam